egguy/addon-grott-beta #7: Install instructions

CK0y0TE, 2023-10-25:
As a HA user I would be very interested as how to install your addon in HA.
After reading the documentation I did not encounter any such instructions. 
Obviously, I and many other growatt users, are not a developper so bear with us.
So far I have tried muppet300, johanmeijers and your documentation and I have encountered many disappointed HA users. Appearantly HA attracts many ignorant IT people?

"If you are using an external MQTT broker, you need to configure the MQTT broker details in the add-on configuration. If you are using the Mosquitto broker add-onn, you can leave the MQTT broker details empty."

No clue what you are saying here. The previously reffered HA mqtt addon was just mentioned to install and configure. "configure the MQTT broker details in the add-on configuration" which one?

But most importantly, I downloaded your latest beta zip. And then what?

candidotsa, 2023-10-26:
Hi. You don't need to install the zip, just use the button add to my home assistant.

egguy, 2023-10-27:
Hello,

This repository is for the individual add-on. As pointed by @candidotsa you are not probably going to use this directly. You can deploy it manually in your HA instance, but it mostly for development purpose.

I can recommend you to use this repository: https://github.com/egguy/ha-addons it contain grott 2.7 and grott 2.8 and it follow a stable release cycle (abeilt a bit slower than the aforementioned ha-addons-edge).

There is instruction on how to add it to your HA installation on the GitHub.

CK0y0TE, 2023-10-27:
Thanks, you guys are wonderful for sharing and commitment to the community.

I was on that path three times already. Problem is grott does not send to mqqt in the way I/mosquitto expects at my py. My tasmota devices do work with the standard HA mqt addon and are reporting just fine with user/passwd set up by me. 

As i mentionned before, the lesser adept, like me and many others, have difficulties following the if-then-else documentation. egguy is refering to “addon” in his instructions but it is not clear if he means the mqt addon in his “module” or the HA mqt  addon. Also, i don’t know docker and thus can’t find the mqtt ini. There’s just so much to learn and so little capacity in my head.

Ill be patient. He, egguy, already mentioned to pickup the docu when he has time.

Gratefull, sincerely,

CK0y0TE, 2023-10-27:
> I have an improved documentation for the add-on, but it's shown only after installation. https://github.com/egguy/addon-grott/blob/main/grott/DOCS.md
>
> I created specifically this add-on to automatically configure the login/password of MQTT and to have a minimal level of configuration. (often just the inverter type + logger is enough) This is why I'm surprised you need to do so many changes.
> The add-on also includes the support for the auto discovery of the devices with MQTT, so they should appear automatically.
>
> Do you have an external mosquito/MQTT server?
>
> For editing the grott file, it's in the doc, you can use the file editor add-on or the vscode add-on included with HA.
>
> Could you please take a look at the doc in the first link (consider it's after the add-on has been installed) and tell me if there are any not clear points? I strive to make the installation easy for the nominal case (HA installed with HASSIO using Mosquito as an Add-on) and possible any other different setup.

Hi,

Thank you a bucket of solar watts!

Let’s see what a noob reads:

- The support for inverters, I remember the first time I entered “MIN” for for my min4200XT. The suggestion came if I ment “min” in lower case. SO I changed that to lower case. not sure what is correct.
- So it is OR specific inverter type OR the multiple inverter config with { "SERIAL": "TYPE", "SERIAL": "TYPE” } (It was already clear to me in the previous docu version but explicit textual expression may read easier for first timers)
- External MQTT broker: "If you are using an external MQTT broker, you need to configure the MQTT broker details in the add-on configuration. If you are using the Mosquitto broker add-onn, you can leave the MQTT broker details empty.” It is this text that totally confused me. No idea which addon you mean. Yours or the HA MQTT addon. The extra text makes it clearer, but still, it doesnt read easily.
- when checking "Enable grott MQTT” the field to enter the host,port,user,passwd is actually the “mqtt” line I suppose? That is below the other options. The order shouldn’t matter much but
(Obviously I don’t need it myself since I use the "HA addon store MQQT” which is working fine.)
And then there is : 
“Verbose”
“Retain”
which both have no effect when not using the Grott Beta 2.8 mqtt?

Next is this line:
"Compatibility with homeassistant-grott <https://github.com/egguy/addon-grott-beta/blob/main/grott-beta/DOCS.md#compatibility-with-homeassistant-grott>” maybe something like this is more fluent? "Compatibility with 'homeassistant-grott <https://github.com/egguy/addon-grott-beta/blob/main/grott-beta/DOCS.md#compatibility-with-homeassistant-grott> component' by muppet3000”

Next :
You can use the add-on and the component at the same time. This will make the deployment of grott and configuration a bit easier.
Makes it easier? Why? what else is there to do and what does the component do? do I actually need it when already using yours? Possible questions from a first time reader.

Next:
You can activate the compatibility mode by checking: Enable grott MQTT in the configuration, and then follow the instruction to install and activate the extension.
Totally confused here. I just unchecked the "Enable grott MQTT” because I already have mqtt addon in homeassistant addon store which is working fine. Do I need it for the previous paragraph about add-on component makes deployment a bit easier?
Which road should one take?
What instructions are you referring to if one enables this option?
Are referring to a “Enable grott MQTT” in the component of muppet3000?

Next:
It is recommended to deactivate the Home Assistant plugin in the configuration. This can lead to duplicate sensors.
In what case is this true? When using HA add-on store mqqt, if using muppet 300 component? (BTW muppet3000 refers to johanmeyers python files to drop in custom_components/grott directory….of which the readme is a general linux implementation documentation I think and not a hassio help?)

Next:
Configuration of grott. Aren't we having a working config by now? Is additional grott.ini config optional or necessary. 

I feel like being a real pain. I realy don’t want to and most of my remarks may be obvious already for the majority of users. But being a retired UNIX IT infrastructure Consultant/Engineer, Oracle DBA Admin, Sun Solaris Technical Consultant I assumed myself to be able ti understand things. Which I don’t. I would like too inspect the log files to see whats going on.

I truly hope I am helping you and others by replying.

THank you for your efforts to include the elderly in this growatt journey and enjoy the HA expierience.

Cheers,

Camiel Kotte

jordanmckoy, 2024-02-29:
HI, I am trying to add this add-on to Home assistant, but I am unable to add the repository to HACS. Am I doing something wrong? 

This is the error I am getting when I put the repo into HACS

[image]

egguy, 2024-03-01:
Hi, this is an add-on, 

you should add it to the add-on configuration of your system ha installation.

jordanmckoy, 2024-03-01:
> add-on configuration

That makes **A LOT** more sense now  😂. Thank you for clarifying that.

typxxi, 2024-03-13:
For all those HA OS users among us that have 
* already setup a grott server for example on pi and have
* also setup mqtt and / or have installed muppet3000 grott integration  

how would your recommand to install the addon grott beta ?

I want to migrate my 7 inverters to my HA OS running on Pi-5 so I assume that I can start with just one and install the addon which will install a grott server instance, but will that be a 2.8.2 or 3 which is capable dealing with multi inverters ?

After the setup I guess I can then move 1 of the inverters to the IP of my HA, but i am not sure and still quite happy about the way muppets integration with the current grott pi server is working.