egguy/addon-grott-beta #9: No shutdown when the value drops to 0

delimedia, 2023-10-28:
I have a small big problem currently.
I have a Growatt MIC 600TL-X inverter.
It turns off the ShineWifi stick when the inverter turns itself off in the evening. 
Unfortunately Grott 2.7 still transmits the last wattage numbers to Growat and also to MQTT.  So it is not recognized that the inverter is off and the value should be 0.

What can I do?

muhkuh666, 2024-01-16:
Have the same issue!

Sorentino1964, 2024-01-30:
Ebenfalls gleiches Problem 
Wechselrichter Growatt 2500 TL

egguy, 2024-02-20:
Hi, I think I found a way to make HA mark the sensor as `Unvailable` after not receiving updates.
@Schelin78 @muhkuh666 Could this work for you?

e.g.
[image]

muhkuh666, 2024-02-20:
How to do so?

egguy, 2024-02-21:
There is a parameter in the MQTT discovery who could be used to make the devices unavailable if there is no new value pushed after a certain time.
So when the inverter shutdown, the push of values cease and the devices would be marked as unavailable 15 minutes after.

I put 15 minutes to prevent transients problems.

If this is what you are looking for, I will push this in the next version of the addon.

Sorentino1964, 2024-02-21:
That's it
That would be very nice if you could implement that

egguy, 2024-03-14:
Hi @Sorentino1964 @muhkuh666 @Schelin78 @delimedia 

Finally! I found a bit of time to update. The addon has been published on the edge repository.

It is possible to stop the regular add-on and install the addon in the edge repository: https://github.com/egguy/ha-addons-edge/
You can copy the configuration of the old addon to the new in YAML mode.

Could you please confirm the addon works for you and does what you want? The expiration of the measures is after 15 minutes (2 default 5m period to prevent jitter + 1 of buffer).

If everything works, I will publish the add-on to the stable repo. Thanks for you feedback

Sorentino1964, 2024-04-04:
Hi @egguy 
Today I installed the new Current version: 0.1.6.
I'll test it for a few days now and then report back.
Thank you again for your good work here.

egguy, 2024-04-04:
Hi, the  1.6 doesn't include the changes for the moment I only pushed it on the edge branch. 

I didn't have time to make a proper release.

Sorentino1964, 2024-04-04:
[image]

Sorentino1964, 2024-04-04:
@egguy 
First test OK👌

egguy, 2024-04-08:
Thank you for the test and feedback. I'm going to push it to the main branch.

closed 2024-06-26 by egguy