egguy/addon-grott-beta #10: Stoped working on 25 October

Schelin78, 2023-11-01:
Shinephone stoped working also
Logg:
T06NNNN  : 
{'decrypt': {'value': 'True'}, 'datalogserial': {'value': 16, 'length': 10, 'type': 'text', 'incl': 'yes'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text'}, 'date': {'value': 136, 'divide': 10}, 'recortype1': {'value': 150, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 154, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 178, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 190, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 194, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 202, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 210, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 214, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 218, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 222, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 230, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 234, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 238, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 254, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 262, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 270, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 278, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 286, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 290, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 294, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 298, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 302, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 306, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 310, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 314, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 318, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 322, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 326, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 330, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 358, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 366, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 374, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 382, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 390, 'length': 4, 'type': 'num', 'divide': 10}}
T05NNNNX  : 
{'decrypt': {'value': 'True'}, 'datalogserial': {'value': 16, 'length': 10, 'type': 'text', 'incl': 'yes'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text'}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 170, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 178, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 182, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 198, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 202, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 206, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 214, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 218, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 222, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 266, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvenergytoday': {'value': 274, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 282, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 290, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 298, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 306, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 314, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 322, 'length': 4, 'type': 'num', 'divide': 10}, 'pvtemperature': {'value': 450, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 466, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 470, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 474, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}}
T06NNNNX  : 
{'decrypt': {'value': 'True'}, 'datalogserial': {'value': 16, 'length': 10, 'type': 'text', 'incl': 'yes'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text'}, 'date': {'value': 136, 'divide': 10}, 'recortype1': {'value': 150, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 154, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 178, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 190, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 194, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 250, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 258, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 262, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 266, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 270, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 278, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 282, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 294, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 298, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 346, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvenergytoday': {'value': 354, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 362, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 370, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 378, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 386, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 394, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 402, 'length': 4, 'type': 'num', 'divide': 10}, 'pvtemperature': {'value': 530, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 546, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 550, 'length': 2, 'type': 'num', 'divide': 10, 'incl': ' [... cut ...]
	extvar:             	 {'ha_mqtt_host': 'core-mosquitto', 'ha_mqtt_port': '1883', 'ha_mqtt_user': 'addons', 'ha_mqtt_password': 'deeRae6Iek9tohb2suu6cudagoh1aedabei4jeNe7Yengeiloo7ieGaivazohvu6', 'ha_mqtt_retain': True}

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott-current
IP :  172.30.33.2 , port :  5279

Schelin78, 2023-11-02:
----------------------
 Add-on version: 0.1.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.1  (armv7 / raspberrypi4)
 Home Assistant Core: 2023.11.0
 Home Assistant Supervisor: 2023.10.1

egguy, 2023-11-06:
Do you see any traffic coming from you stick? If you refresh, you should see new packet coming if the stick configuration is correct.

closed 2024-02-24 by github-actions[bot]