egguy/addon-grott-beta #16: Serial to device mapping

andystewart999, 2023-11-22:
Hi - not really an issue, but possibly helpful for your serial to device mapping logic?

My Growatt Inverter (serial CHAB81905A) isn't recognised, according to the log.  For your reference it's an 5500MTL-S, bought in Australia.

`Growatt data decrypted V2`
`Inverter serial not recognised - using inverter type default`
`Grott HA 0.10.1 - creating CHAB81905A config in HA, 32 to push`

egguy, 2023-11-22:
Hi, 
if you only have one inverter, there is no need to use an inverter mapping.

You probably need to setup the inverter type in the configuration: https://github.com/egguy/addon-grott-beta/blob/main/grott-beta/DOCS.md#specific-inverter-configuration, This part is manual as it's a limitation of grott for the moment.

I don't know the best mapping for this inverter, my guess would be using: `TL3` you can try the other listed.

If you find the correct one, can you tell me which one is working so I can add it to the documentation?

andystewart999, 2023-11-23:
Hmm - manually specifying ANY of the other inverter types resulted in either a crash (see below) or nothing being parsed.

> Traceback (most recent call last):
>   File "/app/grott.py", line 35, in <module>
>     proxy.main(conf)
>   File "/app/grottproxy.py", line 144, in main
>     self.on_recv(conf)
>   File "/app/grottproxy.py", line 238, in on_recv
>     procdata(conf,data)    
>     ^^^^^^^^^^^^^^^^^^^
>   File "/app/grottdata.py", line 220, in procdata
>     for keyword in  conf.recorddict[layout].keys() :
>                     ~~~~~~~~~~~~~~~^^^^^^^^
> KeyError: 'T06NNNNTL3'

With debugging turned on I can see that the record layout for my inverter (or probably more accurately for the datalogger, a Shine WiFi-S) is 'T06NNNN'.

So I've left everything at the defaults for now.  Hopefully this may be of some use, otherwise go ahead and close this off!

Cheers,
Andy

egguy, 2023-11-23:
I've probably gone a bit fast. So it seems to be supported by the default parser: https://github.com/johanmeijer/grott/blob/eafd8f11da54119b89159c5f0fb7a533f068fc6a/README.md?plain=1#L65

The message: `Inverter serial not recognised - using inverter type default` Is normal and is using the generic layout (I know it's a bit unorthodox). If you get all the useful values, it's most likely working as intended.

If it's the case, feel free to close.

For problems with the decoded layout and parsing (visible with the debug mode), the issues should be open here: https://github.com/johanmeijer/grott

closed 2024-02-24 by github-actions[bot]