egguy/addon-grott-beta #21: No contact with api

Gemies, 2024-01-28:
I am having trouble reaching the api. This is the logging of the grott addon beta. Don't know what to do about it.

Grott Beta branch (2.8)
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott Beta branch (2.8)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.1.5
 Home Assistant Supervisor: 2024.01.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[17:35:56] INFO: Preparing to start...
[17:36:02] ERROR: Got unexpected response from the API: Service not enabled
./script.sh: line 230: GROTT_CONFIG_MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

charltondurie, 2024-02-11:
I have also been experiencing the exact same problem over the last few weeks. My Grott is currently down and not able to start as a result. 

Any update on this issue? Please let me know if I can provide any more information

egguy, 2024-02-12:
Hi, could you try to restart the mosquitto addon, then the add-on?

It seems this procedure seems to solve this error. I still don't know the root cause, I will try to find a more definitive fix.

closed 2024-03-22 by github-actions[bot]