egguy/addon-grott-beta #29: LOST IN THE GROWATT - GROTT - HA 'BERMUDA TRIANGLE'

typxxi, 2024-03-13:
General question: 

It is hard to understand what this add-on is clearly for - at least for a beginner like me. Therefore here my story so far.

I have about  8 different growatt inverters and therefore I need solution that are working cause I have to deal with each of the inverter more or less just to get things right like observing my 14 strings and its kWh / kWp performance which growatt is not providing.

I started to install grott not being aware of this add-on and followed the muppet 3000 grott integration path which required a full running grott installation to start with. I struggled a lot with the grott setup cause the documentation is broken in between since the mulitple container architecture has started and the rpi setup or installation needs more to get the right release. 

docker pull ledidobe/grott 
is the command, but not working for you as a pi owner, cause now to get the right version with the full support you need to call
docker pull ledidobe/grott:beta
or
docker pull ledidobe/grott:2.8.3

Took me ages to find these tiny differences out cause I only got a message from grott that i was using the wrong architecture on my pi which I could not believe cause I had followed all the steps and all the links which were there to tell me that rpi is dead you need the new multi architecture what so ever where it again says docker pull ledidobe/grott and NOT docker pull ledidobe/grott:2.8.3

So I was forced to do the manual installation as the alternative for my Pi-3 which should run as the proxy for test purposes.
Then I set it up and could not get it running due to errors from the communication. Grott.ini shows in the mqtt section a user id and a password but I had not read about that i have to set up such user first and what not. But I did it later in HA and then it did not work again eeven though I had removed all the "#" in the mqtt section. I did believe that now everything was right, but it wasn't.
#auth = False 
that is how the line of the grott.ini looks in the example directory but who would guess that you need to switch that to true in case that the grott.ini shows in the next line a user grott and a password grott2020 ?
I would assume that everything is prepared for starting, no, you have to change that to auth = True

For whatever reason it finally ran and so I kept track by installing the grott integration (@muppet3000) which worked great, also the multi inverter integration did not need any additional steps or work, cause I simply moved stick by stick from growatt server to grott and a minute later such inverter appeared in my HA and worked cause soon the entities were created, 34 for each inverter.

This is my journey, took me many days or over a week to get there that I am now happy having 7 inverters running on grott on pi 3 and also all the 34 sensors of each inverter in HA on my Pi-5  running HA OS.

AND NOW I AM on the edge how to integrate or not the GROTT proxy function onto my Pi-5 ?

I had learned so far that I can not run docker containers inside HA OS cause HA OS itself is a kind of container system more or less.

So I had thought, that I simply could use the terminal to install grott and run grott there inside on HA OS but I am not sure cause while investigating I came across this soultion addon-grott-beta which I better can remind as the egguy. 

I assume why it is called this way but beta is usually not the software people wanna start with even though grott is in the latest version only available as beta right now. 
But I got it.

But even I have read the documentation here I am still asking myself what is this really for and if I am right using it.

Is it right that this addon brings basically a full running grott onto a HA OS system replacing the (for HA OS users) currently needed dedicated grott machine on the same HA OS instance ?

If so then it would need a more precise description "Grott for HA OS users" with no need for an extra machine. 
And then it would be time to mention muppet3000 integration as the go to addon to get all the sensors automatically into HA.

Am I right about that ?
Can I replace my current dedicated Pi -3 Grott proxy server (acting for 7 inverters, not just 1 or 2) with addon-grott-beta and will this new addon have the same full functionality ?

Just to let you know: accidentily I did not follow the grott description for multi inverter operations fully, only for the first 2 I had created the yaml files and I had activated this grott.ini line too:    mqttinverterintopic="True"

But: once I switched another stick from growatt server to my grott server the inverters appeared within a  minute later (my sticks report every minute) in HA as new device in my only grott instance and with 34 inverters

Nothing fancy more needed. Move the stick and sensors appeared in HA

Maybe I am wrong but I would like to know that in advance cause right now my Pi-3 server with Chris / @muppet3000 integration is working fine.

But I would of cause prefer a 1 machine solution running HA and under its hood the Grott Addon with the GROTT integration which is confusing for most of the people cause everything seems to be grott. 

I hope that the addon-grott-beta should act like grott under HA and might be called GROTT HA or HA GROTT with a reference that the integration is respronsible for the next step to create all sensors needed by just listenting to the mqtt traffic . 
Would have helped me a lot , saved 3 or 4 days after all the struggles in the beginning.

And I simply had to write it down here to get a real glimpse what each of the component might be needed for and what it requires but for someone with no mqtt series and some usual HA knowledge this is currently a mess for any newbie or better a bermuda triangle.

GROTT it self should say for HA owners that are running HA OS that there are 2 ways 
* setup a dedicated pi or machines acting as grott server (or on a vm / container) or
* install NOT MY GROTT, but the GROTT HA edition called GROTT HA ADDON which will do everything grott is doing 

GROTT HA then should provide people the clear message that this is a kind of a port to run GROTT under HA OS and nothing more.
Finally it would make a lot sense to show those users that they will save a lot of time by installing also muppet3000 grott integration as the bridge from the grott server via mqtt to get the sensors created and running in HA.

Of cause i can be wrong with these assumption who is doing cause I am just a usual / average guy as Klopp once said when he arrived in Liverpool . But I have to start somewhere without risking too much.

Should I install addon-grott-beta on my HA installation and what would I have to change ?
copy the existing grott.ini and then replacing the ip adress in the sticks from the grott installation to the HA server ?

I hope someone can tell me where I am wrong or right. 

Thanks , great job and I will now continue searching for more documentation about this topic before I crash my HA even though I have daily backups on google drive.

egguy, 2024-03-13:
Hi, 

I understand the confusion around the grott ecosystem, how it is difficult to make sense of all the options.There is a bit of overlap in the whole ecosystem,

# parts

I will restate the different parts to help clarify.
- [grott](https://github.com/johanmeijer/grott): Create by @johanmeijer, is a proxy which can decode the growatt stick protocol and forward it to MQTT/influxDB/pvoutput. It supports plugins to add output/functionalities. (This is the interesting part) 
  - The MQTT output is specific to grott, it can be pushed to HA, but you need to [configure each measure individually](https://github.com/johanmeijer/grott/blob/master/examples/Home%20Assistent/mqtt_growatt_eng.yaml)

- Using the plugin architecture, I wrote for a friend a [gott plugin](https://github.com/johanmeijer/grott/blob/master/examples/Home%20Assistent/grott_ha.py) some years ago, to automatically convert the data of grott to the [MQTT format ](https://www.home-assistant.io/integrations/sensor.mqtt/) used in Home Assistant. This communicates directly with **HA and doesn't need an integration**
  - Create HA MQTT configuration topic/mapping for each measure usiStackng native MQTT configuration
  - Reformat MQTT messages to separate MQTT topic (1 per inverted), while keeping the old message.
  - Correct some measures 
  - Now published as a native python lib (https://github.com/egguy/grott-ha-plugin) for easy installation
- My friend, was not being keen on maintaining a container + manual management of the plugin, so, I published an addon based on @Maes152 works https://github.com/Maes152/grott-home-assistant-add-on/.
  - The addon includes grott +  grott HA plugin + autoconfiguration if the mosquitto add-on is present
- @muppet3000, which is the author/maintainer of the official HA growatt integration, created an [HACS integration ](https://github.com/muppet3000/homeassistant-grott/) after the problems with Growatt's server problems. I works in a different but similar way
  - Integrate natively with the HA sensor architecture 
  - Works by listening to grott default MQTT output topic ("energy/growatt/#")
  - Mapping the content of the message to sensor
  - Create calculated sensors to reproduce the one found in growatt dashboard ([custom_components/grott/sensors/sensors_calculated.py](https://github.com/muppet3000/homeassistant-grott/blob/976dd2e4e0405a8894b8e63ec8111d98e6fd19c2/custom_components/grott/sensors/sensors_calculated.py))

# functionality of this add-on

The add-on goal is to allow the user to have an easy installation+configuration of grott and the sensors, if your HA installation supports add-ons (which are just docker container).

As you can see, there is a bit of overlap,  my extension in the add-on also do the publication in Home Assistant of the sensor's values.

I added an option in the add-on to disable the native grott extension and only do the default MQTT output to allow user who want to use the grott HACS  integration to work nicely with my addon. I added a section specifically for this in the documentation: https://github.com/egguy/addon-grott-beta/blob/main/grott-beta/DOCS.md#compatibility-with-homeassistant-grott

# to conclude

This addon also provide a native integration with HA and not only a grott in an add-on. You can reconfigure and add functionality as the grott.ini file is available, and you can disable the native plugin to integrate with homeassistant grott.

As you stated, this is confusing, as the name implies it is only grott. I will take a look to see if I can find a better formulation. In the meantime, I'm going to update the README to make it more clear and dispel the confusion.

Do you have some suggestion of what I could add to help this?

typxxi, 2024-03-23:
A week later and some of the fog has lifted. 
I have arrived which means right now that 7, no 8 inverters are now at least reporting to growatt cause the sun is quite low and the hybrid is not reporting AC production, just DC String production. 

My current first suggestion would be a tiny one here I would not open an issue for cause it is so simple.
[image]

I would add a ".3" for 2.8.3 just to show every one how close you are too grott (which is also 2.8.3) and not to raise question (marks) cause I was wondering since I knew from grott that some major changes had arrived with 2.8.2 and 2.8.3

So I missed the clear grott version which can be key for all those that arrive with an inverter that might make a difference cause I guess a lot of people will again cross read what is going grott wise in general and grott has the bigger community, even though this one is smaller and far more productive. 

And beyond that the Bermuda triangle is there but it is caused due to the high complexity where to start or what you will need.
I hope to find the time in the summer to write something about my use case so that anyone with the same goal can follow those instructions 1:1 where my setup goal was: having grott on a single, but power full HA installation running HA OS

Finally the bermuda triangle is caused by a pile of new terms and requirements if you wanna begin with cause if you simply buy a growatt inverter then you get a good value for the money inverter but that will come to a halt when HA is needed.

In this case you will need to understand 
* the existance of grott as a proxy
* the rerouting and reconfiguration of a sticks IP
* the mqtt add on
* the mqtt (broker) integration (maybe add on explorer) and most likely 
* the grott integration
* and have any clue about what an integration might be and the difference to an add on (also where to find the addon menu in ha which is far more rarely used than integrations) 

And that is what I had known back then 2 years ago (without the grott integration by muppet3000 which did not exist), where your grott ha addon came to me as the last bit and piece cause I first did not understand the use case or why i would need something more after all that mentioned. 

That is a awfull lot. And if you start from the tail as I did being aware of the grott then you might start with reading Grott integration by muppet 3000 which then is referring a lot and in depth to grott basically asking to set up grott first and fully.

But that is the step which had caused me a lot of trouble and headache cause there things went wrong cause i simply could not even install grott due to missing hints how to install a dedicated version like beta or 2.8.3 where the documentation never explained how to get that particular version even though Johann recommands to use 2.8.3 beta in daily life. So you will have to go through all of the grott pain to finaly learn about your grott add on.

With that being said I will close the issue for now and hope that it might help others too. In my case it would have saved me a lot of time (but I then would also have missed lessons learned from basic grott) if i would have taken a different route from the muppet3000 integration cause everyone who is wanting to use his integration will end up in a fully working grott setup, without being aware that the direct way of using your addon would be the easiest way (most likely - at least from my current experience). 

It needs a lot to under stand that you can have grott without having to deal with real grott cause at the end you need your addon grott beta and the grott ha integration (for HA OS users that wanna run the whole stuff on 1 machine). And it still makes me wonder I run grott without grott cause it is the addon which is doing the job.

Great addon, well written documentation and success here.
Hats off, great piece of art.

typxxi, 2024-03-23:
the issue is too complex and complicated to be fixed right now and in 1 step, but might end in more smaller improvements over time. Therefore closed as a whole.

closed 2024-03-23 by typxxi