egguy/addon-grott-beta #32: Grott Wont Start

charltondurie, 2024-03-15:
Since the update 2 days ago (that happens to coincide with a core update), I have not been able to start Grott. See the error log below, specifically the following like:

`./script.sh: line 241: dig: command not found`

I have tried to restart Mosquito Broker then restart Grott but this didn't fix the issue.

Full error log as follows:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott Beta branch (2.8)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 61506dd
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.1
 Home Assistant Supervisor: 2024.03.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[08:05:52] INFO: Preparing to start...
./script.sh: line 241: dig: command not found
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

egguy, 2024-03-16:
I think I have identified the bug. I will publish a remediation soon.

egguy, 2024-03-16:
This should be fixed, could you try the new image once it is published ?

PR #33 (2024-03-16): Add CI with goss + fix error with dig
# Proposed Changes

Add back dig.
Add CI test of the image to prevent this to happen again

## Related Issues

close #32
closed 2024-03-16 by egguy