egguy/addon-grott-beta #47: No MQTT data in HA

InvertedMinor, 2024-06-25:
Despite the settings below no MQTT data is visible in HA?

What am I doing wrong?

Inverter: MOD 6000TL3-XH
MQTT was already installed on HA
Redirected Growatt to HA IP : 192.168.2.33
Installed  Grott Beta branch (2.8) Current version: 0.1.7
Configuration: invertert type : min, Enable grott MQTT 

The IP adddress 172.30.33.0 shown in the logs is not known to me??

`Grott settings:

_Generic:
	version:             	 2.8.3
	verbose:             	 True
	trace:               	 False
	config file:         	 /config/grott/grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 min
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 172.30.33.0
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 addons
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 47.91.67.66
	growattport:         	 5279
_PVOutput:
	pvoutput:            	 False
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 yourapikey
	pvinverters:         	 1
	pvsystemid:          	 systemid1
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
_Extension:
	extension:          	 False
	extname:            	 grottext
	extvar:             	 {'none': 'none'}

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 1ef934e0-grott-beta
IP :  172.30.33.13 , port :  5279 

	 `

` MQTT jsonmsg: 
			 {"device": "XXX", "time": "2024-06-25T05:14:19", "buffered": "yes",
			 "values": {"pvserial": "YYY", "pvstatus": 0, "pvpowerin": 0,
			 "pv1voltage": 4149, "pv1current": 0, "pv1watt": 0, "pv2voltage": 1876,
			 "pv2current": 0, "pv2watt": 0, "pv3voltage": 0, "pv3current": 0, "pv3watt":
			 0, "pv4voltage": 0, "pv4current": 0, "pv4watt": 0, "pvpowerout": 0,
			 "pvfrequentie": 4999, "pvgridvoltage": 2353, "pvgridcurrent": 0,
			 "pvgridpower": 0, "pvgridvoltage2": 2327, "pvgridcurrent2": 0,
			 "pvgridpower2": 0, "pvgridvoltage3": 2317, "pvgridcurrent3": 0,
			 "pvgridpower3": 0, "vacrs": 4052, "vacst": 4020, "vactr": 4043,
			 "ptousertotal": 2513, "ptogridtotal": 0, "ptoloadtotal": 2513,
			 "totworktime": 8671065, "pvenergytoday": 0, "pvenergytotal": 21455,
			 "epvtotal": 22070, "epv1today": 0, "epv1total": 14964, "epv2today": 0,
			 "epv2total": 7106, "epv3today": 0, "epv3total": 0, "etousertoday": 0,
			 "etousertotal": 4436, "etogridtoday": 0, "etogridtotal": 15902,
			 "eloadtoday": 0, "eloadtotal": 10604, "deratingmode": 0, "iso": 4735,
			 "dcir": 0, "dcis": 0, "dcit": 0, "gfci": 4150, "pvtemperature": 301,
			 "pvipmtemperature": 473, "temp3": 292, "temp4": 0, "temp5": 348,
			 "pbusvoltage": 2106, "nbusvoltage": 2042, "ipf": 20000, "realoppercent": 0,
			 "opfullwatt": 60000, "standbyflag": 0, "faultcode": 0, "warningcode": 0,
			 "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0,
			 "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0,
			 "systemfaultword6": 0, "systemfaultword7": 0, "invstartdelaytime": 51,
			 "bdconoffstate": 0, "drycontactstate": 0, "edischrtoday": 0, "edischrtotal":
			 0, "echrtoday": 0, "echrtotal": 0, "eacchrtoday": 0, "eacchrtotal": 0,
			 "priority": 0, "epsfac": 0, "epsvac1": 0, "epsiac1": 0, "epspac1": 0,
			 "epsvac2": 0, "epsiac2": 0, "epspac2": 0, "epsvac3": 0, "epsiac3": 0,
			 "epspac3": 0, "epspac": 0, "loadpercent": 0, "pf": 10000, "dcv": 0,
			 "bdc1_sysstatemode": 0, "bdc1_faultcode": 0, "bdc1_warncode": 0,
			 "bdc1_vbat": 0, "bdc1_ibat": 0, "bdc1_soc": 0, "bdc1_vbus1": 0,
			 "bdc1_vbus2": 0, "bdc1_ibb": 0, "bdc1_illc": 0, "bdc1_tempa": 0,
			 "bdc1_tempb": 0, "bdc1_pdischr": 0, "bdc1_pchr": 0, "bdc1_edischrtotal": 0,
			 "bdc1_echrtotal": 0, "bdc1_flag": 0, "bdc2_sysstatemode": 0,
			 "bdc2_faultcode": 0, "bdc2_warncode": 0, "bdc2_vbat": 0, "bdc2_ibat": 0,
			 "bdc2_soc": 0, "bdc2_vbus1": 0, "bdc2_vbus2": 0, "bdc2_ibb": 0, "bdc2_illc":
			 0, "bdc2_tempa": 0, "bdc2_tempb": 0, "bdc2_pdischr": 0, "bdc2_pchr": 0,
			 "bdc2_edischrtotal": 0, "bdc2_echrtotal": 0, "bdc2_flag": 0, "bms_status":
			 0, "bms_error": 0, "bms_warninfo": 0, "bms_soc": 0, "bms_batteryvolt": 0,
			 "bms_batterycurr": 0, "bms_batterytemp": 0, "bms_maxcurr": 0,
			 "bms_deltavolt": 0, "bms_cyclecnt": 0, "bms_soh": 0, "bms_constantvolt": 0,
			 "bms_bms_info": 0, "bms_packinfo": 0, "bms_usingcap": 0, "bms_fw": 0,
			 "bms_mcuversion": 0, "bms_commtype": 0}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing disabled 

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.13', 5279), raddr=('192.168.2.10', 22467)>
	 - Data less then minimum record length, data not processed
`

egguy, 2024-07-17:
Hi @InvertedMinor,

Did you disable the integrated HA plugin? This section show the plugin is not enabled:
```
_Extension:
extension: False
extname: grottext
extvar: {'none': 'none'}
```

If it is not enabled, no sensor will be automagically created; and then you need to manually add the sensor or add another HACS extension.

InvertedMinor, 2024-07-17:
Thnx, In the meantime I am running grott on a pi because all the numerous instructions and add-ons where just too confusing for me.