egguy/addon-grott-beta #56: Someone have a tutorial to install??

perico85, 2024-07-22:
Hi, i have used the addon till 5 days ago, ans with the changes in growatt server i have lost everything. What i say? I remove everything on my system because i were crazy with why my system stopped to work ans now i need put back again.

In the shinelan setting control web, i had changes the IP of the server to my homeasistant installation (192.168.1.101) but when a install grott addon and grott integration, It doesnt received data
[image]
[image]
[image]

Pd: sorry for my english.

Thanks

typxxi, 2024-07-29:
to make things clear:

I own roughly a dozen growatt inverters and all are running on this addon-grott-beta as HA addon inside HA.
egguy has done a great job to get and keep it running.

Here is 1 HA installation with 8 inverters and roughly 450+ sensors. I am running the addon and I am using muppets Grott HA integration which is creating and providing all the sensors.
[image]

I am not an expert, most of the HA stuff is completely new to me even though i understand all the basics and specialities from my PV installation, DIY batteries, Wallbox and what not. So it is possible.

I guess you simply had a bad starting point or moment, when growatt has caused a lot of troubles with overriding our "ip reroutings via stick to grott". THerefore a lot might have gone wrong for you too cause I had similiar issues and was not aware that it was caused by growatt, cause I had to rewire 8 inverters in those days into a new and second distribuition box, then I had to rewire the whole LAN and set the WLAN up again.

At the end nothing was working anymore regarding my PV sensors and reporting. I had to dig deep to find out that I thought I had lost all the IP settings in the stick.  I assumed that all my inverters had lost the settings cause they were not connected to ac for a week or 10 days.
I had not known about the growatt deleting our grott server ip entries. Therefore a similiar situation for me like you had.

To get back.
You simply have to install the addon,
open the addon and the open the configuration tab.

Be aware that at the first start your configuration will look a bit differently cause I have set unusual option that are usually hidden like the first 2 ones which appear as the switch gblockcmd and the ggrowattip which I will explain later below.

In case of multiple inverters be aware that you can use the following setup - here for example a MIN and a SPH inverter.
BUT: be aware what kind of growatt inverter you own and if that is supported or not and how. Cause some inverters can run but under a different "umbrella". Here is an example which once was true one way or the other: if you owned a mic which back then was not supported yet you could have run that as a "MIN" in this setup or vice versa cause I can not remember how it was back then. 
REMEMBER: Not all growatt inverters are supported fully and officially, lot of workarounds can be found on the web but without a warranty that it will work in your specific case with grott

But you have to start with "default" as inverter type - at least if
you wanna use the following kind of mapping (serialnumber to model type) in inverter mapping.

{"BMK1234567": "MIN", "NPC9876531": "SPH"}
If you simply have just 1 inverter then use the same line for that single inverter and remove ",....."

gmode is proxy
Enable grott mqtt is mandatory
deactivate verbose cause that will give you a quite long logging / protocol
activate mqtt retain is usefull to keep the sensors with values cause without any values your pages and cards would not look good till first   mqtt data have arrived.

The next point is mqtt related and in my case mosqito mqtt has been installed and is running, also the mqtt explorer which is a different topic on its own and is not needed - only in case of troubleshooting quite usefull. 
Again: this is for a HA installation running HA with a mosqito mqtt addon not an additional external dedicated mqtt server.

the last parts are the mqtt data path which should be "/config/grott"

The next point is switch where you can toggle on a lot of settings which are rarely neeeded, but in this case you have to turn that on and start from the top again to look for
ggrowattip and fill in "8.209.71.240"

This will stop growatts resetting of our local grott servers in the wifi sticks.

Once you have entered the ip above you can save these changed settings first.
And then scroll to the bottom and make sure that the network port is "5279".
If not change it and save it again.

You will be asked if you wanna restart grott right now and you choose yes.

Here the screenshots 
[image]

and the lower part

[image]

You do not have to change anything in the command line or grott.ini 

The final step would be to open your browser and login into server.growatt.com 
There you choose your inverter and then scroll down till this headline or section appears:

**my photovoltaic devices** 
[image]

Look carefully at this screen cause in the lower right corner you can see ALL DEVICES and an arrow to the right.

You have to click on that.
[... 4 lines cut ...]
On the lower right part you will find the datalogger settings here
[image]

You click on that and then this dialog appears where you have to set the ip adress to your home assistant servers ip
type in the ip and no / or whatsoever.

Then type in the growatt password in the format "growatt202407dd" where dd stands for the date like today 29.
And then click yes
[image]

From this moment on this growatt inverter will report his data to "192.168.178.2" in the believe he is talking with his home in china. 
HA will receive the messages and due to the configuration Grott will intercept these messages first and then resend them further to the growatt cloudservers in china and also via mqqt addon to your HA grott sensors - in my case the Grott HA integration. 

[... cut ...]  PV production.
The meter then usually stands still during the hours cause the inverter produces what the house consumes.

These were all steps beyond the usual advice to make full backups before and after such changes. 
Of cause you will have to install the GROTT HA integration which I have used and is provided by muppet 3000 which does not require much. 

Tend to avoid playing with ini files cause lot more typo and errors happen then and there. 
The UI provides all you need to set it up right.

typxxi, 2024-07-29:
If you have solved your issue then please close this issue cause here you have gotten a full tutorial at least for grott HA addon running with grott integration - or asked further questions.

perico85, 2024-07-29:
> to make things clear:
>
> I own roughly a dozen growatt inverters and all are running on this addon-grott-beta as HA addon inside HA. egguy has done a great job to get and keep it running.
>
> Here is 1 [...]

thanks you so much, its ok now!! many thanks!

typxxi, 2024-07-30:
As I have learned yesterday. You and I are here on a right now "wrong" beta path which is no longer been updated cause the base was Johan's Grott 2.8.x beta and therefore you can see here the beta in the title.
Johan has released 2.8.3 which is now stable and therefore we are on the wrong path.

Due to the fact that you are also writing on the outdated and dying "branch"  (which has the release number 0.1.7 compared to the most recent stable 0.1.10 if I remember right which reflects the  version of the code for HA or HA specific changes while the underlaying grott is 2.8.3 in the stable branch - I guess this beta one is relying on 2.8.3 beta) I am assuming that you are also on this wrong branch or the no longer maintained one. You might have installed that and came to this page by reading the documentation inside HA and then clicking on a link. That is why I have commented here and was never aware that I am on a dead branch till Etienne = egguy kindly reminded me to switch , when I asked him in return why and where I have to switch.

I had to go into the add on section and open the add on store to find out that there are there 2 versions available:
[image]

ATTENTION: the screenshot is from my own server and as you can see there are 2 versions parallel installed and 1 is not running, that grey one which is not stable.

So follow this guide: install both in parallel, then open the beta one and its configuration.
Then open a new tab, got to the add on view and open the then grey grott stable add on and its configuration which will be empty or in factory reset mode.
Now copy all the  settings step by step carefully and compare those again and save that. 

Last step for me before the actual MOVE : FULL BACKUP to be on the safe side

After that (check if backup is really there) stop the old addon from the main page of the add on.
Once that is done move to the tab with the grott stable configuration and click on NEW START I guess cause I have the german version which looks like this
[image]

Now you are on the stable branch that is maintained and running grott stable 0.1.10 based on grott 2.8.3 

The next big step will be I guess the release of GROTT 3.0 or so where Johan is working on and then Egguy and his "GROTT 3.0 HA Edition" which sound less confusing to me. 

Finally close the issue here cause there is no issue any more that could be solved and then this list would be shorter and looks nicer for anyone who might still arrive here due to other links from the 2.8.3 beta era which might be linking to beta branch.

closed 2024-09-05 by github-actions[bot]